Gator96100/ProxSpace #5: Openocd integration

kevin2008-01, 2018-02-04:
Hi,

Just one question: 
Openocd integration is possible into proxspace ?

Thank you in advance

Gator96100, 2018-02-05:
I'm pretty sure it is possible, but I don't see the point of doing so.
ProxSpace is not meant to be a full collection of development tools, ProxSpace contains just enough tools to compile the Proxmark firmware.
If you want to use OpenOCD within ProxSpace you can always add it yourself by adding precompile builds if they exist or by compiling OpenOCD with ProxSpace yourself.

kevin2008-01, 2018-02-05:
right,

but i'm win user and pm3 bricked with ice fork.
need prox and openocd for repair with pirate.
If not possible, no problem....

Tank you for space for compile.

Gator96100, 2018-08-03:
OpenOCD can now be easily installed with "pacman -S mingw-w64-i686-openocd mingw-w64-x86_64-openocd"

closed 2018-08-03 by Gator96100