Gator96100/ProxSpace #8: errors after install

merlokk, 2018-08-11:
in this file
https://github.com/Gator96100/ProxSpace/blob/master/msys2/etc/post-install/09-proxspace.post
here
pacman --noconfirm -S mingw-w64-i686-gcc mingw-w64-i686-readline mingw-w64-i686-qt5 git make pkg-config
nedds to add "--force" option because clean install cant do this line
because of error: `error: failed to commit transaction (conflicting files)`

Gator96100, 2018-08-11:
I just tried a fresh install and I didn't ran into this error. What package causes this error? Are you sure you had a clean install? 
Adding the "--force" option might hide a damaged ProxSpace environment.

merlokk, 2018-08-12:
mkdir: cannot change permissions of '/dev/shm': Permission denied
mkdir: cannot change permissions of '/dev/mqueue': Permission denied
'C:\Windows\system32\drivers\etc\hosts' -> '/etc/hosts'
'C:\Windows\system32\drivers\etc\protocol' -> '/etc/protocols'
'C:\Windows\system32\drivers\etc\services' -> '/etc/services'
'C:\Windows\system32\drivers\etc\networks' -> '/etc/networks'
gpg: WARNING: using insecure memory!
gpg: please see http://www.gnupg.org/documentation/faqs.html for more information
gpg: /etc/pacman.d/gnupg/trustdb.gpg: trustdb created
gpg: no ultimately trusted keys found
gpg: WARNING: using insecure memory!
gpg: please see http://www.gnupg.org/documentation/faqs.html for more information
gpg: WARNING: using insecure memory!
gpg: please see http://www.gnupg.org/documentation/faqs.html for more information
gpg: Generating pacman keyring master key...
gpg: skipping control `%no-protection' ()
+++++
..+++++
gpg: key 683BFC15 marked as ultimately trusted
gpg: Done
==> Updating trust database...
gpg: WARNING: using insecure memory!
gpg: please see http://www.gnupg.org/documentation/faqs.html for more information
gpg: 3 marginal(s) needed, 1 complete(s) needed, PGP trust model
gpg: depth: 0  valid:   1  signed:   0  trust: 0-, 0q, 0n, 0m, 0f, 1u
==> Appending keys from msys2.gpg...
gpg: WARNING: using insecure memory!
gpg: please see http://www.gnupg.org/documentation/faqs.html for more information
gpg: 3 marginal(s) needed, 1 complete(s) needed, PGP trust model
gpg: depth: 0  valid:   1  signed:   0  trust: 0-, 0q, 0n, 0m, 0f, 1u
==> Locally signing trusted keys in keyring...
  -> Locally signing key D55E7A6D7CE9BA1587C0ACACF40D263ECA25678A...
  -> Locally signing key 123D4D51A1793859C2BE916BBBE514E53E0D0813...
  -> Locally signing key B91BCF3303284BF90CC043CA9F418C233E652008...
  -> Locally signing key 9DD0D4217D75A33B896159E6DA7EF2ABAEEA755C...
==> Importing owner trust values...
gpg: WARNING: using insecure memory!
gpg: please see http://www.gnupg.org/documentation/faqs.html for more information
gpg: setting ownertrust to 4
gpg: setting ownertrust to 4
gpg: setting ownertrust to 4
gpg: inserting ownertrust of 4
==> Updating trust database...
gpg: WARNING: using insecure memory!
gpg: please see http://www.gnupg.org/documentation/faqs.html for more information
gpg: 3 marginal(s) needed, 1 complete(s) needed, PGP trust model
gpg: depth: 0  valid:   1  signed:   4  trust: 0-, 0q, 0n, 0m, 0f, 1u
gpg: depth: 1  valid:   4  signed:   3  trust: 0-, 0q, 0n, 4m, 0f, 0u
gpg: depth: 2  valid:   3  signed:   0  trust: 3-, 0q, 0n, 0m, 0f, 0u
gpg: WARNING: using insecure memory!
gpg: please see http://www.gnupg.org/documentation/faqs.html for more information
gpg: refreshing 8 keys from hkp://pool.sks-keyservers.net
gpg: requesting key 683BFC15 from hkp server pool.sks-keyservers.net
gpg: requesting key CA25678A from hkp server pool.sks-keyservers.net
gpg: requesting key AEEA755C from hkp server pool.sks-keyservers.net
gpg: requesting key 3E0D0813 from hkp server pool.sks-keyservers.net
gpg: requesting key 3E652008 from hkp server pool.sks-keyservers.net
gpg: requesting key A47D45A1 from hkp server pool.sks-keyservers.net
gpg: requesting key 2C51581E from hkp server pool.sks-keyservers.net
gpg: requesting key 4CA56930 from hkp server pool.sks-keyservers.net
gpgkeys: key FD5AAF13558A2A7907C8A033A293E4D5683BFC15 not found on keyserver
gpg: key CA25678A: "Alexey Pavlov (Alexpux) <alexey.pawlow@gmail.com>" 4 new signatures
gpg: key AEEA755C: "Martell Malone (martell) <martellmalone@gmail.com>" 3 new signatures
gpg: key 3E0D0813: "Ray Donnelly (MSYS2 Developer - master key) <mingw.android@gmail.com>" 2 new signatures
gpg: key 3E652008: "Ignacio Casal Quinteiro <icquinteiro@gmail.com>" 2 new signatures
gpg: key A47D45A1: "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" 1 new signature
gpg: key 2C51581E: "Martell Malone (MSYS2 Developer) <martellmalone@gmail.com>" not changed
gpg: key 4CA56930: "Ray Donnelly (MSYS2 Developer) <mingw.android@gmail.com>" 2 new signatures
gpg: Total number processed: 7
gpg:              unchanged: 1
[... 297 lines cut ...]
--> Installing /usr/share/info/libgomp.info.gz ... done
--> Installing /usr/share/info/libidn2.info.gz ... done
--> Installing /usr/share/info/libquadmath.info.gz ... done
--> Installing /usr/share/info/libtasn1.info.gz ... done
--> Installing /usr/share/info/libunistring.info.gz ... done
--> Installing /usr/share/info/m4.info.gz ... done
--> Installing /usr/share/info/mpfr.info.gz ... done
--> Installing /usr/share/info/readline.info.gz ... done
--> Installing /usr/share/info/rluserman.info.gz ... done
--> Installing /usr/share/info/sed.info.gz ... done
--> Installing /usr/share/info/time.info.gz ...install-info: warning: no info dir entry in `/usr/share/info/time.info.gz'
 done
--> Installing /usr/share/info/wget.info.gz ... done
--> Installing /usr/share/info/which.info.gz ... done

###################################################################
#                                                                 #
#                                                                 #
#                   C   A   U   T   I   O   N                     #
#                                                                 #
#                  This is first start of MSYS2.                  #
#       You MUST restart shell to apply necessary actions.        #
#                                                                 #
#                                                                 #
###################################################################

merlokk, 2018-08-12:
resolving dependencies...
looking for conflicting packages...
warning: dependency cycle detected:
warning: mingw-w64-i686-harfbuzz will be installed before its mingw-w64-i686-freetype dependency

Packages (99) bash-4.4.019-3  db-5.3.28-2  expat-2.2.5-1  gdbm-1.11-3  heimdal-1.5.3-9  libgdbm-1.11-3
              mingw-w64-i686-assimp-4.1.0-1  mingw-w64-i686-binutils-2.30-3  mingw-w64-i686-bzip2-1.0.6-6
              mingw-w64-i686-ca-certificates-20170211-2  mingw-w64-i686-crt-git-6.0.0.5136.897300fe-1
              mingw-w64-i686-dbus-1.10.16-1  mingw-w64-i686-expat-2.2.5-1  mingw-w64-i686-fontconfig-2.13.0-1
              mingw-w64-i686-freeglut-3.0.0-4  mingw-w64-i686-freetype-2.9.1-1  mingw-w64-i686-gcc-libs-7.3.0-2
              mingw-w64-i686-gettext-0.19.8.1-3  mingw-w64-i686-giflib-5.1.4-2  mingw-w64-i686-glib2-2.56.1-2
              mingw-w64-i686-gmp-6.1.2-1  mingw-w64-i686-graphite2-1.3.11-1  mingw-w64-i686-harfbuzz-1.7.5-2
              mingw-w64-i686-headers-git-6.0.0.5136.897300fe-1  mingw-w64-i686-icu-61.1-1  mingw-w64-i686-icu-debug-libs-61.1-1
              mingw-w64-i686-isl-0.19-1  mingw-w64-i686-jasper-2.0.14-1  mingw-w64-i686-lcms2-2.9-1
              mingw-w64-i686-libffi-3.2.1-4  mingw-w64-i686-libgcrypt-1.8.2-1  mingw-w64-i686-libgpg-error-1.31-1
              mingw-w64-i686-libiconv-1.15-2  mingw-w64-i686-libjpeg-turbo-1.5.3-1  mingw-w64-i686-libmng-2.0.3-4
              mingw-w64-i686-libpng-1.6.34-1  mingw-w64-i686-libsystre-1.0.1-4  mingw-w64-i686-libtasn1-4.13-1
              mingw-w64-i686-libtiff-4.0.9-1  mingw-w64-i686-libtre-git-r128.6fb7206-2  mingw-w64-i686-libwebp-1.0.0-1
              mingw-w64-i686-libwinpthread-git-6.0.0.5134.2416de71-1  mingw-w64-i686-libxml2-2.9.8-1
              mingw-w64-i686-libxslt-1.1.32-1  mingw-w64-i686-mpc-1.1.0-1  mingw-w64-i686-mpfr-4.0.1-1
              mingw-w64-i686-ncurses-6.1.20180407-1  mingw-w64-i686-openssl-1.0.2.o-1  mingw-w64-i686-p11-kit-0.23.12-1
              mingw-w64-i686-pcre-8.42-1  mingw-w64-i686-pcre2-10.31-1  mingw-w64-i686-python3-3.6.5-1
              mingw-w64-i686-qtbinpatcher-2.2.0-2  mingw-w64-i686-sqlite3-3.23.1-1  mingw-w64-i686-tcl-8.6.8-1
              mingw-w64-i686-termcap-1.3.1-3  mingw-w64-i686-tk-8.6.8-1  mingw-w64-i686-windows-default-manifest-6.4-3
              mingw-w64-i686-wineditline-2.205-1  mingw-w64-i686-winpthreads-git-6.0.0.5134.2416de71-1
              mingw-w64-i686-xpm-nox-4.2.0-5  mingw-w64-i686-xz-5.2.4-1  mingw-w64-i686-zlib-1.2.11-1
              mingw-w64-i686-zziplib-0.13.69-1  openssh-7.6p1-2  perl-5.24.4-1  perl-Authen-SASL-2.16-2
              perl-Convert-BinHex-1.125-1  perl-Encode-Locale-1.05-1  perl-Error-0.17025-1  perl-File-Listing-6.04-2
              perl-HTML-Parser-3.72-2  perl-HTML-Tagset-3.20-2  perl-HTTP-Cookies-6.04-1  perl-HTTP-Daemon-6.01-2
              perl-HTTP-Date-6.02-2  perl-HTTP-Message-6.14-1  perl-HTTP-Negotiate-6.01-2  perl-IO-Socket-SSL-2.054-1
              perl-IO-stringy-2.111-1  perl-LWP-MediaTypes-6.02-2  perl-MIME-tools-5.509-1  perl-MailTools-2.20-1
              perl-Net-HTTP-6.17-1  perl-Net-SMTP-SSL-1.04-1  perl-Net-SSLeay-1.84-1  perl-TermReadKey-2.37-2
              perl-TimeDate-2.30-2  perl-Try-Tiny-0.30-1  perl-URI-1.73-1  perl-WWW-RobotRules-6.02-2  perl-libwww-6.31-2
              vim-8.1.0022-1  git-2.17.1-1  make-4.2.1-1  mingw-w64-i686-gcc-7.3.0-2  mingw-w64-i686-qt5-5.10.1-2
              mingw-w64-i686-readline-7.0.003-1  pkg-config-0.29.2-1

Total Installed Size:  6889.30 MiB

:: Proceed with installation? [Y/n]
(99/99) checking keys in keyring                                               [###########################################] 100%
(99/99) checking package integrity                                             [###########################################] 100%
(99/99) loading package files                                                  [###########################################] 100%
(99/99) checking for file conflicts                                            [###########################################] 100%
error: failed to commit transaction (conflicting files)
bash: /usr/bin/bash.exe exists in filesystem
bash: /usr/bin/bashbug exists in filesystem
bash: /usr/bin/sh.exe exists in filesystem
bash: /usr/share/doc/bash/CHANGES exists in filesystem
bash: /usr/share/doc/bash/COMPAT exists in filesystem
bash: /usr/share/doc/bash/FAQ exists in filesystem
bash: /usr/share/doc/bash/INTRO exists in filesystem
bash: /usr/share/doc/bash/NEWS exists in filesystem
bash: /usr/share/doc/bash/POSIX exists in filesystem
bash: /usr/share/doc/bash/RBASH exists in filesystem
bash: /usr/share/doc/bash/README exists in filesystem
bash: /usr/share/doc/bash/bash.html exists in filesystem
bash: /usr/share/doc/bash/bashref.html exists in filesystem
bash: /usr/share/info/bash.info.gz exists in filesystem
bash: /usr/share/locale/af/LC_MESSAGES/bash.mo exists in filesystem
bash: /usr/share/locale/bg/LC_MESSAGES/bash.mo exists in filesystem
bash: /usr/share/locale/ca/LC_MESSAGES/bash.mo exists in filesystem
bash: /usr/share/locale/cs/LC_MESSAGES/bash.mo exists in filesystem
bash: /usr/share/locale/da/LC_MESSAGES/bash.mo exists in filesystem
bash: /usr/share/locale/de/LC_MESSAGES/bash.mo exists in filesystem
bash: /usr/share/locale/el/LC_MESSAGES/bash.mo exists in filesystem
bash: /usr/share/locale/en@boldquot/LC_MESSAGES/bash.mo exists in filesystem
bash: /usr/share/locale/en@quot/LC_MESSAGES/bash.mo exists in filesystem
[... cut ...]
bash: /usr/share/man/man1/typeset.1.gz exists in filesystem
bash: /usr/share/man/man1/ulimit.1.gz exists in filesystem
bash: /usr/share/man/man1/umask.1.gz exists in filesystem
bash: /usr/share/man/man1/unalias.1.gz exists in filesystem
bash: /usr/share/man/man1/unset.1.gz exists in filesystem
bash: /usr/share/man/man1/until.1.gz exists in filesystem
bash: /usr/share/man/man1/wait.1.gz exists in filesystem
bash: /usr/share/man/man1/while.1.gz exists in filesystem
Errors occurred, no packages were upgraded.
pm3 ~$

merlokk, 2018-08-12:
1st and 2nd start

Gator96100, 2018-08-12:
This error only occurs if you clone this repository, I am not sure why though. If you download the zip everything is working correctly. I did push a fix.

closed 2018-08-12 by Gator96100