Gator96100/ProxSpace #9: A PM3 GUI problem

zywshuai, 2018-09-02:
I wrote the code like this,but i cant get the number behind the "block data:" please help

        -     Case "button_break"
                item.Control = New Button
                item.Control.Text = item.Text
                ToolTip1.SetToolTip(item.Control, item.Tooltip)
                AddHandler item.Control.Click, AddressOf hi
                AddHandler item.Control.Click, AddressOf actionCommand
                'AddHandler hi(), AddressOf actionCommand
                'sector = hi()
                Call hi()
                item.Action &= ("hf mf nested 1" & sector)
                item.Control.Tag = item.Action
                item.Control.SetBounds(0, i * 35, 140, 20)
                item.Control.AutoSize = True
Public Sub hi()
    Dim sector_num() As String
    sector_num = Split(prompt.Text, vbCrLf)
    For i = 0 To UBound(sector_num)
        If Strings.Left(sector_num(i), 11) = "block data:" Then
            sector = Strings.Right(sector_num(i), Strings.Len(sector_num(i)) - 11)
        End If
    Next i
End Sub

proxmark3> hf mf nested 1

Usage:

all sectors: hf mf nested <key A/B> <key (12 hex symbols)> [t,d]

one sector: hf mf nested o <key A/B> <key (12 hex symbols)>

           <target block number> <target key A/B> [t]          
card memory - 0 - MINI(320 bytes), 1 - 1K, 2 - 2K, 4 - 4K, - 1K

t - transfer keys into emulator memory

d - write keys to binary file

  sample1: hf mf nested 1 0 A FFFFFFFFFFFF           
  sample2: hf mf nested 1 0 A FFFFFFFFFFFF t           
  sample3: hf mf nested 1 0 A FFFFFFFFFFFF d           
  sample4: hf mf nested o 0 A FFFFFFFFFFFF 4 A

Gator96100, 2018-09-02:
I am not sure what this has to do with ProxSpace.

closed 2018-09-02 by Gator96100

zywshuai, 2018-09-02:
https://github.com/zywshuai/PM3_GUi   this is my github  can you help me