Gator96100/ProxSpace #13: jansson-devel

merlokk, 2018-11-08:
Hi

I need for compile jansson-devel but i dont see it in mingw(

I need to manually import .h files to project?

merlokk, 2018-11-08:
strange...
https://github.com/Alexpux/MSYS2-packages/blob/master/jansson/PKGBUILD#L70

merlokk, 2018-11-08:
this works
```
pm3 /pm3i/client$ pacman -Ss jansson-devel
msys/jansson-devel 2.11-1 (development)
```

merlokk, 2018-11-08:
`pacman -S jansson-devel`
installed
```
pm3 /pm3i/client$ pacman -Ss jansson
mingw32/mingw-w64-i686-jansson 2.11-1
    A C library for encoding, decoding and manipulating JSON data (mingw-w64)
mingw64/mingw-w64-x86_64-jansson 2.11-1
    A C library for encoding, decoding and manipulating JSON data (mingw-w64)
msys/jansson 2.11-1 (libraries) [installed]
    C library for encoding, decoding and manipulating JSON data (runtime)
msys/jansson-devel 2.11-1 (development) [installed]
    jansson headers and libraries
```
needs to add to proxspace

Gator96100, 2018-11-10:
The package mingw-w64-x86_64-jansson/mingw-w64-i686-jansson is already installed with the latest ProxSpace environment. As far as I know jansson-devel is only for msys and would have no effect on ProxSpace. What feature does require jansson and did you test it with the latest ProxSpace environment?

merlokk, 2018-11-14:
i have to add jansson source to proxmark( because i need path library that links with jansson only that way(

closed 2018-11-14 by merlokk