Gator96100/ProxSpace #14: mingw ansi colors inside pm3 client

iceman1001, 2019-02-21:
One thing that has been on my mind is the strangeness with ansi colors in the mingw based proxspace environment.
Since there is color when we start the mingw env but once we start the proxmark3 client the output stops being able to translate the escape sequences correct.

Does someone have a solution for this?

merlokk, 2019-02-21:
It the same for me. I dont know why.

Gator96100, 2019-02-21:
It might have to do with readline6. Readline7 might work. I'll check it out.

Gator96100, 2019-02-21:
Is there an existing command that should have a colored output?

iceman1001, 2019-02-21:
you can modify  "showBanner" function in client/proxmark.c   (for RRG or iceman based repo)  It has colors

Gator96100, 2019-02-21:
MinGW is a toolchain to compile on Windows natively with no POSIX emulation layer. That means if you want colored output, you need to do it the Windows way. Prior to Windows 10 there were no ansi sequences. The only way is:
```
SetConsoleTextAttribute( GetStdHandle(STD_OUTPUT_HANDLE), FOREGROUND_RED );
printf("Hello World!");
SetConsoleTextAttribute( GetStdHandle(STD_OUTPUT_HANDLE), 0x0f);
```
The reason why there is colored output when you start Proxspace is that the MinGW packages are compiled with msys2 as POSIX emulation layer. Here is a pretty good summery [MSYS2 Shell or MinGW-w64 Shell](https://sourceforge.net/p/msys2/discussion/general/thread/dcf8f4d3/#8473/588e)

There are some possibilities to enable colored output:

- Handle all coloring in the Proxmark source code, with the method described above
- Changing Proxspace from MinGW shell to msys2 shell, this comes at a performance
penalty
- Shipping ProxSpace with a different terminal

iceman1001, 2019-02-21:
Does that setconsoletextattribute work?  I tried a somewhat similar techneque with enableing vt escapes but it didn't work.

iceman1001, 2019-02-21:
and yes,  that did work,  to answer my own question.

iceman1001, 2019-02-21:
mingw is compiled with msys2 and  proxmark3 client is compiled with mingw-toolchain (which doesn't have that ansi stuff)..  It makes more sense now.    However I have a hard time seeing the SetConsoleTextAttribute in a simple  **define** macro way like on linux/apple.

Gator96100, 2019-02-24:
I tried to compile the Proxmark client with msys2 to see what would happen. The problem is that every Makefile and .c/.h file that differentiate between mingw and linux/macOS need to be changed as msys does set other variables that mingw does.  I might look into it again when I have more time available.

iceman1001, 2019-02-24:
how about we drop this one?

closed 2019-02-24 by iceman1001