Gator96100/ProxSpace #15: Proxspace is 18Gb

iceman1001, 2019-04-11:
The current proxspace is 18gb,   without the pm3 repo.
That is massive.  We really need to make it smaller again.  Ppl with laptops has not that large disc always.

The libjansson can be removed again,  and the QT5 must be able to be smaller, I think its 10Gb.

@doegox maybe has some other suggestions.

Gator96100, 2019-04-11:
I will see what I can do.
The libjansson does only take 65KB, but if it is no longer needed I'll remove it.
The pacman cache folder however does take up 4.2 GB if x86 and x64 dependencies are installed, this folder can be safely deleted after the first time installation.
QT5 does take a significant disk space, but there is nothing I can do about it. To make it smaller I would need to build and distribute my own QT5 packages. Maybe I can get rid of the QT5 examples (1.5GB).
With all of the above changes Proxspace for x86 would be 6.2GB, with only x64 dependencies 6.4GB and with x86 and x64 dependencies 11.5GB big.

iceman1001, 2019-04-11:
well,  that is a start.

micolous, 2019-04-14:
Maybe an option to build without QT5?  If I want to test that I haven't broken something on Windows, I normally don't care about the GUI features.

More generally, it may also be possible to build PM3 for Windows using mingw32/64 cross-compilers on Linux, but it would require some adjustment of the Makefiles: I opened https://github.com/Proxmark/proxmark3/issues/816 for this.

Gator96100, 2019-04-26:
The Proxmark client will build just fine without QT5 and is usable, the GUI features are missing of cause.
However I will not remove QT5 from Proxspace.

Gator96100, 2019-04-28:
Proxspace does now cleanup the pacman cache and removes qt5 example folder after install.

closed 2019-04-28 by Gator96100