Gator96100/ProxSpace #16: x64 Environment Breaks Following Package Update

digitalentropy, 2019-08-15:
Steps to reproduce:

1. Download latest version of ProxSpace from repo.
2. Run runme.bat, allow installation to complete.  Restart shell afterwards to complete initial setup.
3. Run runme64.bat, allow installation to complete, restart shell.
4. Run runme.bat and execute "pacman -Syuu" to upgrade packages.  Restart shell.
5. Run runme64.bat and execute "pacman -Syuu", restart shell.

Every time I've done this or any combination of the steps, the 64-bit shell seems to get stuck in an update loop where I get the following errors and warnings every time the 64-bit shell starts:

> warning: option --force is deprecated; use --overwrite instead
>  mingw-w64-x86_64-gcc-libs-8.2.0-3-any                                565.3 KiB  2.44M/s 00:00 [#######################################################] 100%
>  mingw-w64-x86_64-gcc-libs-8.2.0-3-any.sig                            119.0   B   116K/s 00:00 [#######################################################] 100%
> loading packages...
> warning: downgrading package mingw-w64-x86_64-gcc-libs (9.2.0-1 => 8.2.0-3)
> resolving dependencies...
> looking for conflicting packages...
> error: failed to prepare transaction (could not satisfy dependencies)
> :: installing mingw-w64-x86_64-gcc-libs (8.2.0-3) breaks dependency 'mingw-w64-x86_64-gcc-libs=9.2.0-1' required by mingw-w64-x86_64-gcc
> warning: option --force is deprecated; use --overwrite instead
> 
> Cache directory: /var/cache/pacman/pkg/
> :: Do you want to remove ALL files from cache? [Y/n]
> removing all files from cache...
> 
> Database directory: /var/lib/pacman/
> :: Do you want to remove unused repositories? [Y/n]
> removing unused sync repositories...
> warning: option --force is deprecated; use --overwrite instead
>  mingw-w64-x86_64-gcc-8.2.0-3-any                                      28.7 MiB  13.3M/s 00:02 [#######################################################] 100%
>  mingw-w64-x86_64-gcc-8.2.0-3-any.sig                                 119.0   B   116K/s 00:00 [#######################################################] 100%
> loading packages...
> warning: downgrading package mingw-w64-x86_64-gcc (9.2.0-1 => 8.2.0-3)
> resolving dependencies...
> warning: cannot resolve "mingw-w64-x86_64-gcc-libs=8.2.0-3", a dependency of "mingw-w64-x86_64-gcc"
> :: The following package cannot be upgraded due to unresolvable dependencies:
>       mingw-w64-x86_64-gcc
> 
> :: Do you want to skip the above package for this upgrade? [Y/n]
> looking for conflicting packages...
>  there is nothing to do
> warning: option --force is deprecated; use --overwrite instead
> 
> Cache directory: /var/cache/pacman/pkg/
> :: Do you want to remove ALL files from cache? [Y/n]
> removing all files from cache...
> 
> Database directory: /var/lib/pacman/
> :: Do you want to remove unused repositories? [Y/n]
> removing unused sync repositories...
> warning: option --force is deprecated; use --overwrite instead
>  mingw-w64-x86_64-readline-6.3.008-1-any                              327.4 KiB  1741K/s 00:00 [#######################################################] 100%
>  mingw-w64-x86_64-readline-6.3.008-1-any.sig                           96.0   B  93.8K/s 00:00 [#######################################################] 100%
> loading packages...
> warning: downgrading package mingw-w64-x86_64-readline (8.0.000-4 => 6.3.008-1)
> resolving dependencies...
> looking for conflicting packages...
> 
> Packages (1) mingw-w64-x86_64-readline-6.3.008-1
> 
> Total Installed Size:   1.44 MiB
> Net Upgrade Size:      -0.11 MiB
> 
> :: Proceed with installation? [Y/n]
> (1/1) checking keys in keyring                                                                 [#######################################################] 100%
> (1/1) checking package integrity                                                               [#######################################################] 100%
> (1/1) loading package files                                                                    [#######################################################] 100%
> (1/1) checking for file conflicts                                                              [#######################################################] 100%
> (1/1) checking available disk space                                                            [#######################################################] 100%
> :: Processing package changes...
> (1/1) downgrading mingw-w64-x86_64-readline                                                    [#######################################################] 100%
> warning: option --force is deprecated; use --overwrite instead
> 
> Cache directory: /var/cache/pacman/pkg/
> :: Do you want to remove ALL files from cache? [Y/n]
> removing all files from cache...
> 
> Database directory: /var/lib/pacman/
> :: Do you want to remove unused repositories? [Y/n]
> removing unused sync repositories...

Unfortunately I am not familiar enough with MSYS2 to have a good understanding of what the issue is other than an apparent dependency conflict that doesn't seem to resolve.  After the update compiling in the x64 environment also fails with the following error:

> 
> gcc  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../common/zlib -Iuart -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -Werror -g -O3 -mno-ms-bitfields -DHAVE_GUI -DPRESETS -DBMPTST -o reveng/bmptst reveng/bmpbit.c
> ( ./reveng/bmptst && touch reveng/bmptst ) || ( rm -f reveng/bmptst && false )
> reveng: configuration fault.  Update reveng/config.h with these definitions and recompile:
>         #define BMP_BIT   32
>         #define BMP_SUB   16
> make[1]: *** [Makefile:364: reveng/bmptst] Error 1
> make: *** [Makefile:55: client/all] Error 2

Any suggestions?

doegox, 2019-08-15:
reveng compilation error under mingw64 has now been fixed.

Gator96100, 2019-08-15:
I am not sure what this issue has to do with reveng, anyway...

ProxSpace was not really meant to be updated with "pacman -Syuu".
Running "pacman -Syuu" will update mingw-w64-x86_64-gcc to the latest version. Due to a previous issue with gcc on x64 prior of version 8.2 that cause an error for avx acceleration (hardnested), ProxSpace will download and install mingw-w64-x86_64-gcc 8.2.0-3 if any other version of mingw-w64-x86_64-gcc is installed. At the time when I created this fix, the latest stable mingw-w64-x86_64-gcc package version was 7.3.

Msys2 did have a major update and nowadays the latest gcc version is 9.2. Unfortunately for now, you should not update ProxSpace with "pacman -Syuu". 

I will update ProxSpace to include the latest msys2 updates, this could cause other problems so I need to test it excessively. You can expect a major ProxSpace update in the next days.

doegox, 2019-08-15:
Well two issues were reported in this single issue. I reacted on the "After the update compiling in the x64 environment also fails with the following error" part.

Gator96100, 2019-08-15:
I did not know reveng had changes, now it makes sense. I did not look too closely at the compilation errors and just thought it had to do with broken packages.

digitalentropy, 2019-08-15:
Huge thank you to the both of you for the hard efforts made in supporting the code base and the community.

I will do my best to test out changes as time is available and provide functional feedback.

The reason I updated MSYS2 is because in the current default distribution tab auto-complete seems to be broken within the environment unless I update.  If you don’t know what I’m talking about I’ll try to document and open a new issue ticket.

Gator96100, 2019-08-16:
There is now a untested version of the new ProxSpace update avalible: [https://github.com/Gator96100/ProxSpace/tree/v3.2](url)

doegox, 2019-08-18:
I could compile the pm3 on your v3.2 64b without issues (I did it without QT for hdd space reasons).
One minor thing: the terminal window title says still "Select ProxSpace v3.0 - MINGW64" instead of v3.2
May I also suggest to include astyle? (mingw-w64-x86_64-astyle / mingw-w64-i686-astyle)
Thanks !

Gator96100, 2019-08-27:
ProxSpace is now using the latest msys2 packages. I also did upload already setup versions of ProxSpace to the GitHub releases.

closed 2019-08-27 by Gator96100