Gator96100/ProxSpace #19: path and vars

iceman1001, 2019-08-27:
Two issues in one picture.
1.  that user path is not on my computer.  It looks like a path from @Gator96100 
2. the PATH var isnt correctly ended?

[image]

iceman1001, 2019-08-27:
pkg-config tells a story of julius the user...

[image]

Gator96100, 2019-08-27:
That is interesting. I did test it in a VM with a different user and it did work. I will look into it. Maybe it is a better idea to have the packages downloaded in cache and install it when ProxSpace is first run. 

Only QT seems to be affected, that makes sense as QT does some file patching during install.

iceman1001, 2019-08-27:
could be.

Gator96100, 2019-08-27:
I did upload a new version. Now the packages are only in cache and will be installed automatically. After the installation the cache will be cleared. Could you please try again and check if this error still occurs?

closed 2019-11-04 by Gator96100