Gator96100/ProxSpace #21: ProxSpace and GIT installed

sherhannn79, 2019-09-28:
I have a git installed with default settings. 
1. I downloaded ProxSpace v3.2 release ProxSpace-64.7z
2. I run runme64.bat.
3. After the ProxSpace has downloaded all the updates, I run runme64.bat again and I get the following:
[image]

Gator96100, 2019-09-29:
Have a look at [http://proxmark.org/forum/viewtopic.php?id=6573](url)
Did you extract ProxSpace to a location without spaces?

sherhannn79, 2019-09-30:
In the names of the paths I have no spaces, except that the hyphen in the 'ProxSpace-64'. I'll try to remove the hyphen and reinstall.

sherhannn79, 2019-09-30:
After removing the hyphen and reinstalling, the problem is resolved. Thank!

closed 2019-09-30 by sherhannn79