Gator96100/ProxSpace #22: About copying proxspace folder to new computer running error

ewangsoft, 2019-10-17:
I changed my computer, copied the "proxspace" folder to the new computer, ran "runme64", and prompted "unknown + user @ desktop-bdf7vmj mingw64" error. What should I do?

digitalentropy, 2019-10-18:
Is the folder path the same on the new computer?

Per the instructions, once you set up ProxSpace for the first time you cannot change the path without breaking functionality.

ewangsoft, 2019-10-18:
> Is the folder path the same on the new computer?
> 
> Per the instructions, once you set up ProxSpace for the first time you cannot change the path without breaking functionality.

The original folder is in partition D, now in partition E.
That's not OK?

digitalentropy, 2019-10-18:
To my knowledge it must be the same exact path.

You can probably manually change the path in the configuration files but I haven’t dug into it to see what needs to be adjusted.

ewangsoft, 2019-10-18:
I put it back in the previous position and still prompt "unknown + user @ desktop-bdf7vmj mingw64 ~"

iceman1001, 2019-10-18:
...proxspace is quite allergic to spaces in the folderpath leading to it.  Make sure you don't have any..
```
this will error...
c:\program data\myuser\proxspace
```

Gator96100, 2019-10-18:
ProxSpace/msys2 is not suited for path changes, however it can be done. The unknown user problem is a result of an error in msys2/user_setup.sh most likely because a space or special character in the path and not because you moved ProxSpace. When ProxSpace is moved, every package needs to be reinstalled (although you might get away with just reinstalling qt5, not tested!!!). For reinstalling every package, a script like this can be used: 
```
#! /bin/sh

for pkg in $(pacman -Q | cut -d' ' -f1); do
    pacman -S --noconfirm $pkg
done
```
As you need to reinstall everything you should only do it if there is a good reason to use an existing ProxSpace setup. The reinstallation will most likely take longer and use more bandwidth than a new installation of ProxSpace. The package cache is cleared after the first installation and pacman needs to download every package again.

ewangsoft, 2019-10-19:
> ProxSpace/msys2 is not suited for path changes, however it can be done. The unknown user problem is a result of an error in msys2/user_setup.sh most likely because a space or special character in [...]

Thank you. The whole package has been downloaded and reinstalled. It may take less time than repairing the installation.

closed 2019-10-19 by ewangsoft