Gator96100/ProxSpace #23: ProxSpace Slow to Start

digitalentropy, 2019-10-26:
I have consistently observed that with v3.2 ProxSpace is VERY slow to start while the Internet is connected.  Whatever update it's checking on, it often takes forever, and it seems to be much faster to disconnect from the Internet each time I start it and then re-connect once the prompt has fully loaded.

What is being checked online each time, and why does it take so long?  I have done several clean installs of ProxSpace on several different Windows 10 systems and they all seem to behave the same way.

digitalentropy, 2019-10-26:
Today I seem to be having fewer issues, but I am still unable to identify the source of the delay whenever it happens, other than it being related to an update check.  I would note that the problem seems to happen on all sorts of different networks, and I have consistently observed other users having the same problem during workshops.

Gator96100, 2019-10-26:
After the initial update, ProxSpace will not check for updates over the internet, it will only check the cache for updates. Try removing this line `yes | pacman -Syuu` in ProxSpace/msys2/etc/post-install/09-proxspace.post:55

digitalentropy, 2019-10-27:
That line was actually on line 57 for me (right before “fi)

Removing it resulted in the following error upon start:

“bash: /etc/post-install/09-proxspace.post: line 58: syntax error near unexpected token `fi'
bash: /etc/post-install/09-proxspace.post: line 58: `fi'”

The only reason I assumed it was checking for some sort of update is because nearly every time I started it, it would appear to hang for several minutes before displaying:

“:: Starting core system upgrade...
there is nothing to do
:: Starting full system upgrade...
there is nothing to do”

The fastest way for me to break out of this was to disconnect my WiFi, and in fact I would not have the same delay if starting it while disconnected from the Internet.  This even persisted after I did a clean install with latest git version because I considered maybe something was corrupted.

So all this begs the question: What is the cause of the inconsistency?

If I hadn’t witnessed the same behavior with my own eyes on other systems I would have chalked it up to something unique to me.

Gator96100, 2019-10-27:
I see, remove the `else` before that line too. To my knowledge does `pacman -Syuu` should not access the internet, but I could be wrong

digitalentropy, 2019-10-27:
Ah yes, dumb error on my part.  That did seem to resolve the error.

If it continues to happen maybe I’ll fire up Wireshark and see if I can nail down what’s happening.

digitalentropy, 2019-11-02:
Just a quick update:

I observed the same issue again, even after making the changes to 09-proxspace.post.  Whatever is happening, it's happening before any characters are printed to the console.

I might have to actually do some digging and see what, if anything, is calling out.

Gator96100, 2019-11-04:
Does this still happen when you delete 09-proxspace.post? (Delete it after the initial setup of ProxSpace)

digitalentropy, 2019-11-11:
So far deleting the file has seemed to significantly reduce start time.  I have not yet observed the delay since removing the 09-proxspace.post file but I will continue to monitor.

Gator96100, 2019-11-11:
Replace the 09-proxspace.post with the code below and post the output from ProxSpace. If possible, monitor the CPU usage, IO usage and Network usage during the start up of ProxSpace to check for possible hardware bottlenecks.
```
check_install () { 
	startPacman2=`date +%s`
	pacman -Q $1
	endPacman2=`date +%s`
	runtimePacman2=$((endPacman2-startPacman2))
	echo "ProxSpace: check_install1($1): $runtimePacman2"
	if [ $? == 1 ]; then
		startPacman3=`date +%s`
		pacman --noconfirm --overwrite='*' -S $1
		endPacman3=`date +%s`
		runtimePacman3=$((endPacman3-startPacman3))
		echo "ProxSpace: check_install2($1): $runtimePacman3"
	fi
}

check_install_web () { 
	pacman -Q $1 | grep $2
	if [ $? == 1 ]; then
		pacman --noconfirm --overwrite='*' -U $3
	fi
}

clean_86 () { 
	startClean86=`date +%s`
	rm -rf /mingw32/share/qt5/examples
	endClean86=`date +%s`
	runtimeClean86=$((endClean86-startClean86))
	echo "ProxSpace: clean86 rm: $runtimeClean86"
	startPacman5=`date +%s`
	pacman --noconfirm --overwrite='*' -Scc
	endPacman5=`date +%s`
	runtimePacman5=$((endPacman5-startPacman5))
	echo "ProxSpace: clean64 pacman: $runtimePacman5"
}

clean_64 () { 
	startClean64=`date +%s`
	rm -rf /mingw64/share/qt5/examples
	endClean64=`date +%s`
	runtimeClean64=$((endClean64-startClean64))
	echo "ProxSpace: clean64 rm: $runtimeClean64"
	startPacman4=`date +%s`
	pacman --noconfirm --overwrite='*' -Scc
	endPacman4=`date +%s`
	runtimePacman4=$((endPacman4-startPacman4))
	echo "ProxSpace: clean64 pacman: $runtimePacman4"
}

setup_proxspace ()
{ 
  if [ "$MSYSTEM" == "MINGW64" ]; then
	check_install mingw-w64-x86_64-gcc
	check_install mingw-w64-x86_64-readline
	check_install git
	check_install make
	check_install pkg-config
	check_install mingw-w64-x86_64-qt5
	clean_64
  fi
  
  if [ "$MSYSTEM" == "MINGW32" ]; then
	check_install mingw-w64-i686-readline
	check_install mingw-w64-i686-gcc
	check_install git
	check_install make
	check_install pkg-config
	check_install mingw-w64-i686-qt5
	clean_86	
  fi
}
startProxSpace=`date +%s`

export LANG=en_US.UTF-8
echo "ProxSpace: start"
if [ "$MAYBE_FIRST_START" = "false" ]; then
	echo "ProxSpace: MAYBE_FIRST_START = FALSE"
	startPacman1=`date +%s`
	yes | pacman -Su
	endPacman1=`date +%s`
	runtimePacman1=$((endPacman1-startPacman1))
	echo "ProxSpace: packages updated in: $runtimePacman1"
	startSetup=`date +%s`
	setup_proxspace
	endSetup=`date +%s`
	runtimeSetup=$((endSetup-startSetup))
	echo "ProxSpace: setup_proxspace finished in: $runtimeSetup"
else
	echo "ProxSpace: MAYBE_FIRST_START = TRUE"
	yes | pacman -Syuu
	echo "ProxSpace: packages updated"
fi

endProxSpace=`date +%s`
runtimeProxSpace=$((endProxSpace-startProxSpace))
echo "ProxSpace: runtimeProxSpace:$runtimeProxSpace"
```

Gator96100, 2020-02-02:
Starting with ProxSpace 3.3 this should no longer be an issue.

closed 2020-02-02 by Gator96100