Gator96100/ProxSpace #24: toolchain installation failing since today

doegox, 2019-11-04:
Hi

Our appveyor is failing with the same error since today, see example here:
https://ci.appveyor.com/project/RfidResearchGroup/proxmark3/builds/28592066
And today I got someone at a NFC training getting the same error when he tried to install a fresh ProxSpace.
Error in Appveyor is "make not found" and is caused by the compilation toolchain not being installed properly.
Sorry I can't copy/paste the exact initial error but you should be able to easily reproduce it as it failed both on Appveyor and someone's Windows with the same error...

doegox, 2019-11-04:
I suspect some changes in the upstream mingw repository

Gator96100, 2019-11-04:
What is the output from `pacman -S make`?

Gator96100, 2019-11-04:
Ok I think I found the problem:
```
error: package 'mingw-w64-x86_64-gcc' was not found
error: invalid option '--force'
error: package 'mingw-w64-x86_64-readline' was not found
error: invalid option '--force'
error: package 'git' was not found
error: invalid option '--force'
error: package 'make' was not found
error: invalid option '--force'
error: package 'pkg-config' was not found
error: invalid option '--force'
error: package 'mingw-w64-x86_64-qt5' was not found
error: invalid option '--force'
error: invalid option '--force'
```
The pacman --force option is no longer a warning instead it is now removed.
I did push a fix, please try if it works now.

doegox, 2019-11-04:
Great thanks!
I'll test now in a VM, then a new release will probably be required ;)

doegox, 2019-11-04:
Yes, everything works fine with your latest commit!

Gator96100, 2019-11-05:
Great news! I did publish a new release. It was time to update the cached packages anyway.

closed 2019-11-05 by Gator96100