Gator96100/ProxSpace #25: Windows 10 fails

Darkcast, 2019-11-28:
comments
Comments
@Darkcast
 running on windwos 10 and on the lastest version of both proxmark and proxspace64
when I try to I make install
and I get the following error (see bold below)

pm3 ~$ make clean && make all

Platform name: Proxmark3 rdv4
PLATFORM: PM3RDV4
Platform extras: No extra selected
Included options: SMARTCARD FLASH LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode: LF_SAMYRUN
[] MAKE client/clean
[] MAKE bootrom/clean
[] MAKE fpga_compress/clean
[] MAKE armsrc/clean
[] MAKE recovery/clean
[] MAKE mfkey/clean
[*] MAKE nonce2key/clean

Platform name: Proxmark3 rdv4
PLATFORM: PM3RDV4
Platform extras: No extra selected
Included options: SMARTCARD FLASH LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode: LF_SAMYRUN
[*] MAKE client/all
[-] CC proxmark3.c
[-] CC uart/uart_posix.c
[-] CC uart/uart_win32.c
[-] CC ui.c
[-] CC ../common/commonutil.c
[-] CC util.c
[-] CC ../common/util_posix.c
[-] CC scandir.c
[-] CC ../common/crc16.c
[-] CC comms.c
[-] CC ../common/crapto1/crapto1.c
[-] CC ../common/crapto1/crypto1.c
[-] CC mifare/mfkey.c
[-] CC tea.c
[-] CC fido/additional_ca.c
[-] CC fido/cose.c
[-] CC fido/cbortools.c
[-] CC fido/fidocore.c
[-] CC crypto/asn1dump.c
[-] CC crypto/libpcrypto.c
[-] CC crypto/asn1utils.c
[-] CC cliparser/argtable3.c
[-] CC cliparser/cliparser.c
[-] CC loclass/cipher.c
[-] CC loclass/cipherutils.c
[-] CC loclass/ikeys.c
[-] CC loclass/elite_crack.c
[-] CC fileutils.c
[-] CC whereami.c
[-] CC mifare/mifarehost.c
[-] CC ../common/parity.c
[-] CC ../common/crc.c
[-] CC ../common/crc64.c
[-] CC ../common/legic_prng.c
[-] CC ../common/iso15693tools.c
[-] CC prng.c
[-] CC graph.c
[-] CC cmddata.c
[-] CC ../common/lfdemod.c
[-] CC emv/crypto_polarssl.c
[-] CC emv/crypto.c
[-] CC emv/emv_pk.c
[-] CC emv/emv_pki.c
[-] CC emv/emv_pki_priv.c
[-] CC emv/test/cryptotest.c
[-] CC emv/apduinfo.c
[-] CC emv/dump.c
[-] CC emv/tlv.c
[-] CC emv/emv_tags.c
[-] CC emv/dol.c
[-] CC emv/emvjson.c
[-] CC emv/emvcore.c
[-] CC emv/test/crypto_test.c
[-] CC emv/test/sda_test.c
[-] CC emv/test/dda_test.c
[-] CC emv/test/cda_test.c
[-] CC emv/cmdemv.c
[-] CC emv/emv_roca.c
[-] CC mifare/mifare4.c
[-] CC mifare/mad.c
[-] CC mifare/ndef.c
[-] CC cmdanalyse.c
[-] CC cmdhf.c
[-] CC cmdhflist.c
[-] CC aidsearch.c
[-] CC cmdhf14a.c
[-] CC cmdhf14b.c
[-] CC cmdhf15.c
[-] CC cmdhfepa.c
[-] CC cmdhflegic.c
[-] CC cmdhficlass.c
[-] CC cmdhfmf.c
[-] CC cmdhfmfu.c
[-] CC cmdhfmfp.c
[-] CC cmdhfmfhard.c
[-] CC hardnested/hardnested_bruteforce.c
[-] CC cmdhfmfdes.c
[-] CC cmdhftopaz.c
[-] CC cmdhffido.c
[-] CC cmdhffelica.c
[-] CC cmdhfthinfilm.c
[-] CC cmdhw.c
[-] CC cmdlf.c
[-] CC cmdlfawid.c
[-] CC cmdlfcotag.c
[-] CC cmdlfem4x.c
[-] CC cmdlffdx.c
[-] CC cmdlfguard.c
[-] CC cmdlfgallagher.c
[-] CC cmdlfhid.c
[-] CC cmdlfhitag.c
[-] CC cmdlfio.c
[-] CC cmdlfindala.c
[-] CC cmdlfjablotron.c
[-] CC cmdlfkeri.c
[-] CC cmdlfnexwatch.c
[-] CC cmdlfnedap.c
[-] CC cmdlfnoralsy.c
[-] CC cmdlfpac.c
[-] CC cmdlfparadox.c
[-] CC cmdlfpcf7931.c
[-] CC cmdlfpresco.c
[-] CC cmdlfpyramid.c
[-] CC cmdlfsecurakey.c
[-] CC cmdlft55xx.c
[-] CC cmdlfti.c
[-] CC cmdlfviking.c
[-] CC cmdlfvisa2000.c
[-] CC cmdlfmotorola.c
[-] CC cmdtrace.c
[-] CC cmdflashmem.c
[-] CC cmdflashmemspiffs.c
[-] CC cmdsmartcard.c
[-] CC cmdusart.c
[-] CC cmdwiegand.c
[-] CC cmdparser.c
[-] CC cmdmain.c
[-] CC pm3_binlib.c
[-] CC scripting.c
[-] CC cmdscript.c
[-] CC pm3_bitlib.c
[-] CC cmdcrc.c
[-] CC ../common/bucketsort.c
[-] CC flash.c
[-] CC wiegand_formats.c
[-] CC wiegand_formatutils.c
[-] UIC ui/ui_overlays.h
[-] CXX proxgui.cpp
In file included from proxgui.cpp:14:0:
*proxguiqt.h:68:5: error: 'Ui' does not name a type; did you mean 'IUri'?
Ui::Form opsController;
^~
IUri
make[1]: *** [Makefile:441: obj/proxgui.o] Error 1
make: *** [Makefile:98: client/all] Error 2
pm3

Gator96100, 2019-11-28:
This looks more like a Proxmark issue a not a ProxSpace issue.

I can to compile the latest Proxmark3 rdv4 client with no problem on a fresh ProxSpace environment. 
Did you test if the same error occurs when using a different PC?
I see you are in no subdirectory when executing make all. Did you get the latest source code using the git command within ProxSpace?

iceman1001, 2019-11-29:
The proxguiqt.ccp fails,  hinting on a QT problem.   Not proxmark..

Gator96100, 2019-12-19:
This is most likely an improper QT setup. 
Did you test if the same error occurs when using a different PC/VM?
Did you try to reinstall ProxSpace?
Have/Had you QT installed on your PC that might cause problems with the QT provided in ProxSpace? 
Please send the output from following commands within the ProxSpace shell
```
echo $PATH
pacman -Q mingw-w64-x86_64-qt5
```

Gator96100, 2020-02-02:
Does this error still occur with the latest ProxSpace version? If so please send the output of the ps-info command.

closed 2020-03-09 by Gator96100