Gator96100/ProxSpace #26: Problem about building the fpga_lf.bit and fpga_hf.bit

feifeilanniao, 2019-12-17:
**Describe the bug**
A clear and concise description of what the bug is.
I'd like to build the fpga_lf.bit and fpga_hf.bit by myself, but it seem that some pack missing.
**To Reproduce**
Steps to reproduce the behavior:
1.git clone 'https://github.com/RfidResearchGroup/proxmark3.git' to '/pm3'
2. Go to 'proxmark3-master/fpga'
3. Execute 'make'

**Screenshots**
[image]

feifeilanniao, 2019-12-17:
i just want to refresh the image, i try to the 'make clean && make all',and flash the 'fullimage.elf' into the device, after the 'hw version' command, some messages show that nothing change to the image, due to nothing change to the fpga_lf.bit and fpga_hf.bit?

messages return after ‘hw version’ command like below
     "LF image built for 2s30vq100 on 2019-07-31 at 15:57:16",
    " HF image built for 2s30vq100 on 2018-09-03 at 21:40:23"

iceman1001, 2019-12-17:
you will need to use the xilinik buildchain in order to compile fpga part.  This isn't inside the Proxspace env.

feifeilanniao, 2019-12-17:
ise webpack? i have a try. thank you @iceman1001

Gator96100, 2020-02-02:
I did look into including it with Proxspace, however the Xilinx ISE WebPACK does not allow redistribution.

closed 2020-02-02 by Gator96100