Gator96100/ProxSpace #29: getting error arm-none-eabi-gcc: No such file or directory during make clean && make all

hornyd, 2020-01-24:
After runme64.bat and it completes without error
at prompt, i follow the steps mentioned in the manual

During the make clean && make all, it always stop with errors:

[-] CC amiibo.c
[-] CC drbg.c
[-] CC keygen.c
[=] AR libamiibo.a
[=] GEN lualibs/pm3_cmd.lua
[=] GEN lualibs/mfc_default_keys.lua
[=] LD proxmark3
[*] MAKE bootrom/all
[=] GEN version.c
[-] CC version.c
make[1]: arm-none-eabi-gcc: No such file or directory
make[1]: *** [../common_arm/Makefile.common:74: obj/version.o] Error 127
make: *** [Makefile:92: bootrom/all] Error 2

proxspace@DESKTOP-S0RCVSS MINGW64 ~
#

Any suggestion what it is going wrong?

Gator96100, 2020-01-24:
Have a look at: [http://proxmark.org/forum/viewtopic.php?id=6573](http://proxmark.org/forum/viewtopic.php?id=6573)

> proxspace@DESKTOP-S0RCVSS MINGW64 ~

This indicates that your path for ProxSpace might have invalid characters.

hornyd, 2020-01-27:
Hi,
Thank you for the link, I followed the thread and also the documentation. I put under C:\proxmark\proxspace and also under d:\. I unzipped the file under that folder and runme64.bit (I also tried the runme.bat). I tried with admin and without admin. the admin I run with right click run as admin. but I still get the same error and even once the pacman error (with no exit so the windows stop and no further happen. after re-opening the runme64.bat, it goes immediatly to pm3 prompt.
I disabled my firewall and the anti virus because I saw wget.exe was recognized as virus

I also did with colleague laptop and we are having exactly the same. We are using company laptop so Entreprise Win 10. I tried on my personal laptop also with Win 10 Home

I am a bit desesperate because I could make it work in the past and I am not sure whats going on.

Is there any log or debugging something can help?

Gator96100, 2020-01-27:
Probably related to #28.
After opening the runme, check if msys2/etc/post-install/09-proxspace.post is missing. Just copy it from the zip to the correct folder for now. I am working on a fix.

hornyd, 2020-01-27:
this file is now present (as you mentioned), i am getting now the following:

warning: /etc/pacman.conf installed as /etc/pacman.conf.pacnew
warning: terminate MSYS2 without returning to shell and check for updates again
warning: for example close your terminal window instead of calling exit

Gator96100, 2020-01-27:
That is the expected msys2 update message. Close the cmd windows and run runme again.

hornyd, 2020-01-27:
OK i understand, also after that step, 09-proxspace.post is gone while it was there initially.
I placed back and now when I launch, it continues the installation

hornyd, 2020-01-27:
Super brillant, it works indeed and I could compile proxmarkIII
Will close this issue for now.

closed 2020-01-27 by hornyd