Gator96100/ProxSpace #32: Different size between compiled files

Windslab, 2020-03-03:
After secusefull compiling I got bigger size of the firmware compare to [Gator96100 distribs](https://drive.google.com/drive/folders/1uX9RtYGinuFrpHybu4xq_BE3HrobI20e)

There is my compiled firmware is flasing good but some thimes I got an jumping err of USB Discriptor
My collom is left, right is Gator96100:

[image]

My installation of ProxSpace is clear and can't be any error

PS: Gator96100, please add to auto build: eml; bin; html; json to pack it into zip files.(it already in .gitignore of proxmark repo)

Windslab, 2020-03-04:
Do anybody alive here?

Gator96100, 2020-03-08:
I am just very busy at the moment, so please be patient. 
Different sizes could mean many things. Please post the output from the `ps-info` command, run `declare > info.txt` and upload the created info.txt. Compare your gcc-arm-none-eabi folder with the one in the release .7z

Windslab, 2020-03-08:
> I am just very busy at the moment, so please be patient.
> Different sizes could mean many things. Please post the output from the `ps-info` command, run `declare > info.txt` and upload the created info.txt. Compare your gcc-arm-none-eabi folder with the one in the release .7z

Please find the all data at the link below:
https://we.tl/t-RzYiRFse9P

Gator96100, 2020-03-08:
Try extracting the ProxSpace-64.7z directly on the C:/ drive to keep the path short.

Windslab, 2020-03-08:
> Try extracting the ProxSpace-64.7z directly on the C:/ drive to keep the path short.

Okay, where I should to get ProxSpace-64.7z?

Gator96100, 2020-03-08:
It is the release file: [https://github.com/Gator96100/ProxSpace/releases/download/v3.3/ProxSpace-64.7z](url)

Windslab, 2020-03-08:
After run have this in loop:
```
--2020-03-08 22:58:51--  (try:11)  http://repo.msys2.org/mingw/i686/mingw32.db
Connecting to repo.msys2.org (repo.msys2.org)|45.59.69.178|:80... failed: Connection timed out.
Retrying.
```
```
error: could not lock database: File exists
  if you're sure a package manager is not already
  running, you can remove /var/lib/pacman/db.lck
```
Something went wrong. But what I really know that you should kindly check all ProxSpace. Will happy to help you. I have no problems with my machine that's I can totaly garantide.

Gator96100, 2020-03-08:
This is due to the repo.msys2.org server being down. At the moment is seems to work again. It is not a loop, it just takes 20 tries for each package before switching to another mirror. Please try again.

Windslab, 2020-03-14:
UPD: I have checked all error possibilities and what I see that's after linux compilation the same situation. That why you should check both all with Iceman. Seems you use old wersion of QT, not current.

Windslab, 2020-03-14:
PS: Please give me an invite to repo

Gator96100, 2020-03-15:
QT should not affect the firmware at all, only the client. You cannot compare the Windows builds with linux builds. 
If you wish to make changes to the ProxSpace repository, make a pull request.

Windslab, 2020-03-15:
> QT should not affect the firmware at all, only the client.

So why we have different size of output files if we are use the same repo?

> You cannot compare the Windows builds with linux builds.

Why not? Problem only with firmware for flashing. It is the same at all platforms or not? :D

> If you wish to make changes to the ProxSpace repository, make a pull request.

I get it

Gator96100, 2020-03-15:
> So why we have different size of output files if we are use the same repo?

That is what this issue is for. 

> Why not? Problem only with firmware for flashing. It is the same at all platforms or not? :D

The toolchains are different between Windows and Linux, as well as code segments are different between the operating systems.

iceman1001, 2020-03-15:
Yes,  If you compile on *nix, you get a smaller file.  The compiler (arm-none-eabi-*) is most likely newer and makes better optimization on *nix systems than what is available for Mingw.

Windslab, 2020-03-15:
[image]
My fresh compilation from Windows

Downloaded from Gator google drive:
[image]

Windslab, 2020-03-16:
After compilation at WSL situation the same! Yes you can blame my Proxmark, but then my question is: why it working perfect with Gator builds. After compiling all functional works good but USB discriptor error makes me cry. What's going on?

I filling that you have old QT's or compiler. Can you share your compilation spaces?

[image]

[image]

Gator96100, 2020-03-20:
I did create a powershell script for you to test. If necessary, enable script executions on your system with: 
`Set-ExecutionPolicy unrestricted`

It should setup a ProxSpace environment on C:\ProxSpace-master.  Please compare the generated fullimage.elf file to the one on my GitHub. QT or any other msys2 package has no effect on the firmware.

Ps-test.ps1:
```
if (!([Security.Principal.WindowsPrincipal][Security.Principal.WindowsIdentity]::GetCurrent()).IsInRole([Security.Principal.WindowsBuiltInRole] "Administrator")) { Start-Process powershell.exe "-NoProfile -ExecutionPolicy Bypass -File `"$PSCommandPath`"" -Verb RunAs; exit }
Write-Host "Downloading ProxSpace:"
[Net.ServicePointManager]::SecurityProtocol = [Net.SecurityProtocolType]::Tls12
Invoke-WebRequest "https://github.com/Gator96100/ProxSpace/archive/master.zip" -outfile "C:\proxspace.zip"
Write-Host "Extracting ProxSpace:"
Expand-Archive -LiteralPath "C:\proxspace.zip" -DestinationPath "C:\"
Remove-Item "C:\proxspace.zip"
Write-Host "1st start of ProxSpace"
Start-Process "cmd.exe"  "/c ""cd /D C:\ProxSpace-master && runme64.bat"""
Start-Sleep -s 5
Wait-Process -Name "cmd" -ErrorAction SilentlyContinue
Start-Sleep -s 5
Write-Host "2nd start of ProxSpace"
Start-Process "cmd.exe"  "/c ""cd /D C:\ProxSpace-master && call msys2\ps\setup.cmd && msys2\msys2_shell.cmd -mingw64 -defterm -no-start -c ""git clone https://github.com/Proxmark/proxmark3.git && exit"""""
Start-Sleep -s 5
Wait-Process -Name "cmd" -ErrorAction SilentlyContinue
Start-Sleep -s 5
Stop-Process -Name "bash" -Force -ErrorAction SilentlyContinue
Stop-Process -Name "gpg-agent" -Force -ErrorAction SilentlyContinue
Stop-Process -Name "dirmngr" -Force -ErrorAction SilentlyContinue
Start-Sleep -s 2
Write-Host "3rd start of ProxSpace"
Start-Process "cmd.exe"  "/c ""cd /D C:\ProxSpace-master && call msys2\ps\setup.cmd && echo C:\ProxSpace-master\builds /builds ntfs noacl 0 0 >>msys2/etc/fstab && call msys2\msys2_shell.cmd -mingw64 -defterm -no-start -c /ps/autobuild.sh"""
```

Windslab, 2020-03-20:
It in the infinity pause stage...

[image]

Gator96100, 2020-03-20:
Yes close the window, but don't close the powershell window. After closing it should reopen.

Windslab, 2020-03-20:
It doesn't go next...

Gator96100, 2020-03-20:
Try killing the cmd.exe

Windslab, 2020-03-20:
I ran commands from your script by hands. Please take a look what I got:

[image]

Windslab, 2020-03-20:
Okay I see that's official size repository is the same

Windslab, 2020-03-20:
Yeah as I guess isuues can be with Iceman repo. Just see what hapened with x64 builds:

My build:

[image]

Your build:

[image]

[image]

Windslab, 2020-03-20:
But what I see with compare of image contains directly PLC instructions:

[image]

Like a being compiling process includes some of additional information. Thats only one explane.

Gator96100, 2020-03-21:
Are you sure that you are comparing the correct files? My latest fullimage.elf (rdv40-64-20200320-8ef3f3438435157faf613d1b816c83c08d81915c) is 266 KB (272 788 Bytes) big, the same size as yours.

Gator96100, 2020-03-21:
You are comparing my rrg_other-64 build (PLATFORM=PM3OTHER) to your rdv40-64 build (PLATFORM=PM3RDV4). Of cause there is a size differance. Your proplems might come from using a RDV4 firmware on a device that is not RDV4.

closed 2020-03-21 by Gator96100