Gator96100/ProxSpace #38: build error

jiangyi1985, 2020-07-20:
```
 [-b BaseAddress] [-o Offset] [-s DllSuffix] [-T FileList | -] [-4|-8] [-p] [-t] [-i] [-v]
pm3 ~$ make clean && make all                                                                                                                                                                                                                make -C client  clean
make[1]: Entering directory '/pm3/client'
rm -f proxmark3 flasher fpga_compress  proxmark3.exe  flasher.exe  fpga_compress.exe obj/uart_posix.o obj/uart_win32.o obj/util.o obj/util_posix.o obj/ui.o obj/comms.o obj/cmdsmartcard.o obj/pcsc.o obj/crapto1/crapto1.o obj/crapto1/crypto1.o obj/crypto/libpcrypto.o obj/crypto/asn1utils.o obj/crypto/asn1dump.o obj/cliparser/argtable3.o obj/cliparser/cliparser.o obj/fido/additional_ca.o obj/fido/cose.o obj/fido/cbortools.o obj/fido/fidocore.o obj/mifare/mfkey.o obj/loclass/cipher.o obj/loclass/cipherutils.o obj/loclass/ikeys.o obj/loclass/elite_crack.o obj/loclass/fileutils.o obj/whereami.o obj/mifare/mifarehost.o obj/mifare/mifare4.o obj/mifare/mad.o obj/mifare/ndef.o obj/parity.o obj/crc.o obj/crc16.o obj/crc64.o obj/iso14443crc.o obj/iso15693tools.o obj/graph.o obj/cmddata.o obj/lfdemod.o obj/emv/crypto_polarssl.o obj/emv/crypto.o obj/emv/emv_pk.o obj/emv/emv_pki.o obj/emv/emv_pki_priv.o obj/emv/test/cryptotest.o obj/emv/apduinfo.o obj/emv/dump.o obj/emv/tlv.o obj/emv/emv_tags.o obj/emv/dol.o obj/emv/emvjson.o obj/emv/emvcore.o obj/emv/test/crypto_test.o obj/emv/test/sda_test.o obj/emv/test/dda_test.o obj/emv/test/cda_test.o obj/emv/cmdemv.o obj/emv/emv_roca.o obj/cmdhf.o obj/cmdhflist.o obj/cmdhf14a.o obj/cmdhf14b.o obj/cmdhf15.o obj/cmdhfepa.o obj/cmdhflegic.o obj/cmdhficlass.o obj/cmdhfmf.o obj/cmdhfmfp.o obj/cmdhfmfu.o obj/cmdhfmfhard.o obj/hardnested/hardnested_bruteforce.o obj/cmdhftopaz.o obj/cmdhffido.o obj/cmdhw.o obj/cmdlf.o obj/cmdlfawid.o obj/cmdlfcotag.o obj/cmdlfem4x.o obj/cmdlffdx.o obj/cmdlfgproxii.o obj/hidcardformatutils.o obj/hidcardformats.o obj/cmdlfhid.o obj/cmdlfhitag.o obj/cmdlfio.o obj/cmdlfindala.o obj/cmdlfjablotron.o obj/cmdlfnexwatch.o obj/cmdlfnoralsy.o obj/cmdlfparadox.o obj/cmdlfpcf7931.o obj/cmdlfpresco.o obj/cmdlfpyramid.o obj/cmdlfsecurakey.o obj/cmdlft55xx.o obj/cmdlfti.o obj/cmdlfviking.o obj/cmdlfvisa2000.o obj/cmdlfpac.o obj/cmdparser.o obj/cmdmain.o obj/scripting.o obj/cmdscript.o obj/pm3_binlib.o obj/pm3_bitlib.o obj/protocols.o obj/taginfo.o  obj/deflate.o obj/adler32.o obj/trees.o obj/zutil.o obj/inflate.o obj/inffast.o obj/inftrees.o obj/guidummy.o obj/hardnested/hardnested_bf_core_NOSIMD.o obj/hardnested/hardnested_bitarray_core_NOSIMD.o obj/hardnested/hardnested_bf_core_MMX.o obj/hardnested/hardnested_bitarray_core_MMX.o obj/hardnested/hardnested_bf_core_SSE2.o obj/hardnested/hardnested_bitarray_core_SSE2.o obj/hardnested/hardnested_bf_core_AVX.o obj/hardnested/hardnested_bitarray_core_AVX.o obj/hardnested/hardnested_bf_core_AVX2.o obj/hardnested/hardnested_bitarray_core_AVX2.o obj/*.o *.moc.cpp ui/ui_overlays.h lualibs/usb_cmd.lua
cd ../liblua && make clean
make[2]: Entering directory '/pm3/liblua'
rm -f liblua.a lua luac lapi.o lcode.o lctype.o ldebug.o ldo.o ldump.o lfunc.o lgc.o llex.o lmem.o lobject.o lopcodes.o lparser.o lstate.o lstring.o ltable.o ltm.o lundump.o lvm.o lzio.o lauxlib.o lbaselib.o lbitlib.o lcorolib.o ldblib.o liolib.o lmathlib.o loslib.o lstrlib.o ltablib.o loadlib.o linit.o  lua.o luac.o
make[2]: Leaving directory '/pm3/liblua'
cd ./jansson && make clean
make[2]: Entering directory '/pm3/client/jansson'
rm -f dump.o error.o hashtable.o hashtable_seed.o load.o memory.o pack_unpack.o strbuffer.o strconv.o utf.o path.o value.o
rm -f libjansson.a
make[2]: Leaving directory '/pm3/client/jansson'
cd ../common/mbedtls && make clean
make[2]: Entering directory '/pm3/common/mbedtls'
rm -f aes.o asn1parse.o asn1write.o base64.o bignum.o ctr_drbg.o entropy_poll.o entropy.o error.o timing.o ecp.o ecp_curves.o certs.o camellia.o blowfish.o cipher_wrap.o cipher.o cmac.o des.o ecdsa.o md.o md_wrap.o md5.o oid.o pem.o arc4.o pk.o pk_wrap.o pkwrite.o pkcs5.o pkcs12.o pkparse.o platform.o platform_util.o rsa.o rsa_internal.o sha1.o sha256.o sha512.o threading.o x509.o x509_crl.o x509_crt.o
rm -f libmbedtls.a
make[2]: Leaving directory '/pm3/common/mbedtls'
cd ./tinycbor && make clean
make[2]: Entering directory '/pm3/client/tinycbor'
rm -f cborencoder.o cborencoder_close_container_checked.o cborerrorstrings.o cborparser.o cborparser_dup_string.o cborpretty.o cbortojson.o cborvalidation.o
rm -f tinycbor.a
make[2]: Leaving directory '/pm3/client/tinycbor'
make[1]: Leaving directory '/pm3/client'
make -C bootrom  clean
make[1]: Entering directory '/pm3/bootrom'
rm -rf obj/*.o
rm -rf obj/*.elf
rm -rf obj/*.s19
rm -rf obj/*.map
rm -rf obj/*.d
rm -rf version.c
make[1]: Leaving directory '/pm3/bootrom'
make -C armsrc  clean
make[1]: Entering directory '/pm3/armsrc'
rm -rf obj/*.o
rm -rf obj/*.elf
rm -rf obj/*.s19
rm -rf obj/*.map
rm -rf obj/*.d
rm -rf obj/*.z
rm -rf obj/*.bin
rm -rf version.c fpga_version_info.c
make[1]: Leaving directory '/pm3/armsrc'
make -C recovery  clean
make[1]: Entering directory '/pm3/recovery'
rm -f bootrom.bin fullimage.bin proxmark3_recovery.bin
make[1]: Leaving directory '/pm3/recovery'
make -C tools/mfkey  clean
make[1]: Entering directory '/pm3/tools/mfkey'
rm -f crypto1.o crapto1.o parity.o util_posix.o mfkey.o mfkey32 mfkey64  mfkey32.exe  mfkey64.exe
make[1]: Leaving directory '/pm3/tools/mfkey'
make -C client  all
make[1]: Entering directory '/pm3/client'
Compiling liblua, using platform mingw
cd ../liblua && make mingw
make[2]: Entering directory '/pm3/liblua'
make "LUA_A=lua52.dll" "LUA_T=lua.exe" \
"AR=gcc -shared -o" "RANLIB=strip --strip-unneeded" \
"SYSCFLAGS=-DLUA_BUILD_AS_DLL" "SYSLIBS=" "SYSLDFLAGS=-s" lua.exe
make[3]: Entering directory '/pm3/liblua'
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lua.o lua.c
make[3]: gcc: No such file or directory
make[3]: *** [<builtin>: lua.o] Error 127
make[3]: Leaving directory '/pm3/liblua'
make[2]: *** [Makefile:112: mingw] Error 2
make[2]: Leaving directory '/pm3/liblua'
make[1]: *** [Makefile:312: lua_build] Error 2
make[1]: Leaving directory '/pm3/client'
make: *** [Makefile:33: client/all] Error 2
pm3 ~$
```

Looks like gcc command can not find file with relative path?

Gator96100, 2020-07-20:
There was a problem installing gcc, have you tried reinstalling ProxSpace?
What is the output from the ps-info command?

jiangyi1985, 2020-07-21:
ps-info result:
```
pm3 ~$ ps-info
ProxSpace info:
Version: 3.4
Arch: x86_64
OS: Microsoft Windows 10 Enterprise              64-bit          10.0.18363
CPU: Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz
Ram: 17130278912
cat: /ps/installed32.txt: No such file or directory
Installed32:
Installed64: 2020-07-21 23:08:26
OLDPWD: /d/Repos_others/ProxSpace
Path: /gcc-arm-none-eabi/bin:/ps/bin:/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/
arm-none-eabi-gcc: 9.2.1
/ps/bin/ps-info: line 17: gcc: command not found
gcc:
/ps/bin/ps-info: line 18: git: command not found
git:
/ps/bin/ps-info: line 19: make: command not found
make:
/ps/bin/ps-info: line 20: pkg-config: command not found
pkg-config:
/ps/bin/ps-info: line 21: qmake: command not found
Qt:
```

I tried to reinstall, but I got many "key xxxxxx is unknown" errors
```
 [-b BaseAddress] [-o Offset] [-s DllSuffix] [-T FileList | -] [-4|-8] [-p] [-t] [-i] [-v]
'C:\WINDOWS\system32\drivers\etc\hosts' -> '/etc/hosts'
'C:\WINDOWS\system32\drivers\etc\protocol' -> '/etc/protocols'
'C:\WINDOWS\system32\drivers\etc\services' -> '/etc/services'
'C:\WINDOWS\system32\drivers\etc\networks' -> '/etc/networks'
warning: Public keyring not found; have you run 'pacman-key --init'?
error: mingw32: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: mingw64: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: msys: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
warning: Public keyring not found; have you run 'pacman-key --init'?
error: mingw32: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: mingw64: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: msys: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: duplicated database entry 'bash'
error: could not open file /var/lib/pacman/local/db-5.3.28-2/desc: No such file or directory
error: could not open file /var/lib/pacman/local/expat-2.2.9-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/gdbm-1.18.1-2/desc: No such file or directory
error: could not open file /var/lib/pacman/local/git-2.26.2-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/heimdal-7.7.0-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/libgdbm-1.18.1-2/desc: No such file or directory
error: could not open file /var/lib/pacman/local/make-4.3-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/mingw-w64-i686-jansson-2.12-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/mingw-w64-i686-lua-5.3.5-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/mingw-w64-x86_64-jansson-2.12-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/mingw-w64-x86_64-lua-5.3.5-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/openssh-8.2p1-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/p7zip-16.02-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/perl-5.30.2-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/perl-Authen-SASL-2.16-2/desc: No such file or directory
error: could not open file /var/lib/pacman/local/perl-Convert-BinHex-1.125-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/perl-Encode-Locale-1.05-1/desc: No such file or directory
error: could not open file /var/lib/pacman/local/perl-Error-0.17029-1/desc: No such file or directory
[... 440 lines cut ...]
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: mingw64: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: msys: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
warning: Public keyring not found; have you run 'pacman-key --init'?
error: mingw32: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: mingw64: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely
error: msys: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" is unknown
error: key "AD351C50AE085775EB59333B5F92EFC1A47D45A1" could not be looked up remotely

Cache directory: /var/cache/pacman/pkg/
:: Do you want to remove ALL files from cache? [Y/n]
removing all files from cache...

Database directory: /var/lib/pacman/
:: Do you want to remove unused repositories? [Y/n]
removing unused sync repositories...
pm3 ~$
```

Gator96100, 2020-07-21:
Your ProxSpace environment is not executing the post-install files.

`[-b BaseAddress] [-o Offset] [-s DllSuffix] [-T FileList | -] [-4|-8] [-p] [-t] [-i] [-v]`
This does usually happen when you git clone ProxSpace. Did you git clone ProxSpace rather than download the zip file? This could cause this Problem.

jiangyi1985, 2020-07-21:
Yes I did git clone master branch.

closed 2020-07-21 by jiangyi1985