Gator96100/ProxSpace #39: autobuild.bat -  mingw32: key is unknown / database 'mingw64' is not valid

GregorBiswanger, 2020-07-27:
Hello community,

Unfortunately I'm not so lucky with the installation of ProxSpace on Windows 10.

After restarting runme64.bat and starting autobuild.bat I get line-by-line error messages with "mingw32: key is unknown / database 'mingw64' is not valid"

Even if the bash was loaded at the end, he does not know the "make" command. I installed over Chocolatey, but it doesn't go through without problems. I get the same errors as when I ran the bat files.

I downloaded / cloned ProxSpace via master zip, I also tried the last release zip. The same problem for everyone.

Output from autobuild.bat:

error: mingw32: key "4A6129F4E4B84AE46ED7F635628F528CF3053E04" is unknown
error: key "4A6129F4E4B84AE46ED7F635628F528CF3053E04" could not be looked up remotely
error: mingw64: key "4A6129F4E4B84AE46ED7F635628F528CF3053E04" is unknown
error: key "4A6129F4E4B84AE46ED7F635628F528CF3053E04" could not be looked up remotely
error: package 'zip' was not found

error: mingw32: key "4A6129F4E4B84AE46ED7F635628F528CF3053E04" is unknown
error: key "4A6129F4E4B84AE46ED7F635628F528CF3053E04" could not be looked up remotely
error: mingw64: key "4A6129F4E4B84AE46ED7F635628F528CF3053E04" is unknown
error: key "4A6129F4E4B84AE46ED7F635628F528CF3053E04" could not be looked up remotely
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
Processing: armsrc-32

/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: bootrom-32
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: client-32
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: common-32
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: common_arm-32
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: common_fpga-32
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: doc-32
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: docker-32
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
[... 111 lines cut ...]
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: recovery-64
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: tools-64
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found
Processing: traces-64
/ps/autobuild.sh: line 25: git: command not found
/ps/autobuild.sh: line 26: git: command not found
/ps/autobuild.sh: line 27: git: command not found
/ps/autobuild.sh: line 30: make: command not found
/ps/autobuild.sh: line 31: make: command not found

Does anyone have a solution?

Gator96100, 2020-07-27:
I do have the same problem with a fresh install. It looks like it has to do with [2020-06-29 - new packagers](https://www.msys2.org/news/#2020-06-29-new-packagers)

Gator96100, 2020-07-28:
A new release will follow soon.

closed 2020-07-28 by Gator96100

Gator96100, 2020-08-11:
All ProxSpace version prior to v3.5 will no longer work as there was a keyring changes.