Gator96100/ProxSpace #41: Err with pgp keys

Windslab, 2020-08-11:
[image]

Gator96100, 2020-08-11:
I need a little more information.
Please post the full output.
Are you running the latest ProxSpace version?

Windslab, 2020-08-11:
> I need a little more information.
> Please post the full output.
> Are you running the latest ProxSpace version?

That's full.
ps-repiar and all of the ps-* does't work, with the same err.
I think something with a date, that's similar to 5 month Proxspace as it's has installed.
How do I can clear the all chache?

Gator96100, 2020-08-11:
All ProxSpace version prior to v3.5 will no longer work as there was a keyring changes.

Windslab, 2020-08-11:
So do I need reinstall it?

Gator96100, 2020-08-11:
You need to download the latest version and install it again.

closed 2020-08-11 by Gator96100

Windslab, 2020-08-11:
Many Thanks,
You can do your product better next time, by releasing new! :)

Gator96100, 2020-08-11:
Maybe next time you verify if it is still an issue before creating one.