Gator96100/ProxSpace #46: Still unable to Flash bootrom or Firmware

bad4dayz, 2021-06-29:
I completed all the instructions for Windows installation, to the point where I type in the Flasher command.
./client/flasher comx -b ./bootrom/obj/bootrom.elf 
I also tried 
client/flasher comx -b bootrom/obj/bootrom.elf 
either way I get ... no such file or directory

[flash.txt](https://github.com/Gator96100/ProxSpace/files/6730494/flash.txt)
[flash.txt](https://github.com/Gator96100/ProxSpace/files/6730496/flash.txt)

I get to this point and I'm unable to continue... ICEMAN, i followed your instructions and tried previous versions to no avail. I tried to follow Chrissy Morgan's video, and got to the same point and not Flash... additionally, I couldn't find a FLASHER.EXE file either.

Any assistance would be appricated

Thank you

Gator96100, 2021-06-29:
The dedicated flasher.exe was removed some time ago, unfortunately most guides are quite outdated. Flashing is now done with the `pm3-flash-bootrom` and `pm3-flash-fullimage` scripts. 

For more information look at: https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md#flash-the-bootrom--fullimage

closed 2021-06-29 by Gator96100

iceman1001, 2021-06-29:
the videos becomes out-dated quite fast,  but the guides / notes should all be up-to-date and well proven to work.

bad4dayz, 2021-06-30:
do I have to assign the port, it just sits and states waiting for Proxmark3 to appear, yes it's plugged in, and i loaded the drivers to Com 3 and it's presenting in Device Manager as Proxmark3 (com3) ... Thanks all

Gator96100, 2021-06-30:
Did you try to specify the port by yourself? When coming from an unknown firmware the auto detection might not work properly.

bad4dayz, 2021-07-01:
Yes I specified com3 to all flash files,(they were com5) when I run pm3-flash-all; i get -waiting for Proxmark3 to appear... what could I be missing? thank you in advance

bad4dayz, 2021-07-01:
Truth be told everyone, I'm 60 years old, and was programing back with FORTRAN77 so it's been a while for me, and I purchased the ProxMark3 RDV4 because I was told it was the best for cloning... all I want to do i cone FOB's for condo's in my area, and This is the best I'm told. But I can't get past the Flash issue.  I have changed all the flash*.bat files to specially identify com3, and device manager identifies a Proxmark3 on com3 but i get the error... -waiting for proxmark to appear... I just am looking forward to doing FOB Cloning, I followed Chrissy's video, but i can't get past Flashing, same with all the previous documentation, is the a straight forward 60 year old goof proof method :P Thanks all, sorry if this sounds pedantic. I just am looking for a lot of assistance, Thanks

bad4dayz, 2021-07-01:
erased all. re pulled zip and compiled... worked perfectly; got clone from GIT, perfect, copied files PM3 director;  make clean && make all .. worked perfectly...... ran  ./pm3-flash-all and got this...  :(
[image]  

Proxmark3 is on com3 and attached
Gator96100, i found your video posted Nov 16 2020 and was able to follow it... but I stopped at the flash again

bad4dayz, 2021-07-01:
device manager
[image]

Gator96100, 2021-07-01:
It’s an RDV4? In this case `pm3-flash-all` should work out of the box. 
You can specify the com port with following commands:
```
./client/proxmark3 COM3 --flash --unlock-bootloader --image bootrom/obj/bootrom.elf
./client/proxmark3 COM3 --flash --unlock-bootloader --image armsrc/obj/fullimage.elf
````

iceman1001, 2021-07-01:
Do please remember that this is an community effort.   It doesn't sound that its a issue with this Proxspace env.
We have the discord server for asking questions and getting help from others in the community.
You find it here:   https://discord.gg/BNKMqZcK

or ask questions on the Proxmark3 forum,  however its quite silent nowadays. 

With that said,   try, what is called,  the button trick.
unplug yr device,  press and hold the pm3 button,  start the bootrom flash,   don't let go of the button until its finished.
then flash fullimage normally after this.

bad4dayz, 2021-07-02:
ICEMAN.... your suggestion worked Perfect.... thank you  :)
[image]