Gator96100/ProxSpace #47: Updating git w/ perm issues

F9Alejandro, 2021-07-09:
Updating git with having permissions issues will break pacman and prevent use of proxspace as it will be entirely broken. Had to try and update git to have a version that has remote-https due to git no longer allowing http.

Just wanted to put this out there more for future pay attention and make sure everything is not marked as blocked in file properties for the 7z file.

Gator96100, 2021-07-09:
It is known that Git does break ProxSpace. Git will change line breaks and permissions of files. That is the reason why all guides will tell you to download the master zip or the releases instead of cloning the repo. There is also a startup check that should give and error when using git.

I’m not entirely sure what exactly the problem was you were facing. Did you try to update an existing ProxSpace installation using git? Did you clone the repository and the installation failed?

F9Alejandro, 2021-07-09:
Just updating existing git due to it not having remote-https. It was a fairly new version of proxspace as well, about 3 commits behind when I grabbed it. The repo attempted clone but since there is no support for http anymore it failed. The permissions issues are more likely my fault than anything else. When I get home (or have a moment), I can attempt a download and "install" just unblocking the 7z file should help somewhat. Then continue with the install like normal.

Gator96100, 2021-07-09:
Updating an existing ProxSpace install is not supported. Think of ProxSpace as installer, once installed updating it is kind of tricky. There is the `pacman -Syuu` command to update the packages and the `ps-upgrade` command to update the ProxSpace installation, although that might not always work. The cleanest way up update ProxSpace is to do a fresh install.

closed 2021-07-09 by Gator96100