Gator96100/ProxSpace #48: Uninstalled the usb driver

SnickerZ123, 2021-09-27:
I uninstalled the usb driver as I couldn't find the device and it was in the hidden section.
I can no longer find it at all, how do I get it back?

Gator96100, 2021-09-28:
What Windows version do you use?

On Windows 10 the driver will be installed automatically. If you uninstall the driver unplugging the pm3 and plugging it back in will reinstall the driver. Sometimes using a different usb port will also cause the driver to be reinstalled.

For older Windows versions you have to manually install the driver, which can be found in the repo you are using.

SnickerZ123, 2021-09-28:
done thanks

SnickerZ123, 2021-09-28:
hey, any chance you could show me how this works?
I'd pay you to teach me, I've encountered some errors and I can't find anything to fix them

Gator96100, 2021-09-28:
I don’t offer paid support. I recommend you join the Discord server, where you can get a lot of help https://discord.gg/QfPvGFRQxH

If you are not happy with the community support, @iceman1001 does offer paid support at patreon: https://www.patreon.com/bePatron?u=16606719

closed 2021-09-28 by Gator96100